yellobyte/DacESP32 #1: Compilation error on a fresh install of Arduino IDE (1.8.19)

Fabgar1, 2022-09-15:
I changed windows on my notebook and did a new installation of Arduino IDE, now my applications running the library dacesp32 are not compiling anymore, below I tried to compile the example "https://github.com/yellobyte/DacESP32/tree/main/examples/outputAnalogDigitalMixed" and this error came with the compilation:

Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\user1\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\user1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\user1\Documentos\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none -ide-version=10819 -build-path C:\Users\user1\AppData\Local\Temp\arduino_build_167513 -warnings=default -build-cache C:\Users\user1\AppData\Local\Temp\arduino_cache_195834 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mkspiffs.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0 -verbose C:\Users\user1\OneDrive\Documentos\Arduino\sketch_sep15a\sketch_sep15a\sketch_sep15a.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\user1\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\user1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\user1\Documentos\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none -ide-version=10819 -build-path C:\Users\user1\AppData\Local\Temp\arduino_build_167513 -warnings=default -build-cache C:\Users\user1\AppData\Local\Temp\arduino_cache_195834 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mkspiffs.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=C:\Users\user1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0 -verbose C:\Users\user1\OneDrive\Documentos\Arduino\sketch_sep15a\sketch_sep15a\sketch_sep15a.ino

Using board 'esp32' from platform in folder: C:\Users\user1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4

Using core 'esp32' from platform in folder: C:\Users\user1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4

cmd /c if exist "C:\\Users\\user1\\OneDrive\\Documentos\\Arduino\\sketch_sep15a\\sketch_sep15a\\partitions.csv" COPY /y "C:\\Users\\user1\\OneDrive\\Documentos\\Arduino\\sketch_sep15a\\sketch_sep15a\\partitions.csv" "C:\\Users\\user1\\AppData\\Local\\Temp\\arduino_build_167513\\partitions.csv"

cmd /c if not exist "C:\\Users\\user1\\AppData\\Local\\Temp\\arduino_build_167513\\partitions.csv" if exist "C:\\Users\\user1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.4\\variants\\esp32\\partitions.csv" COPY "C:\\Users\\user1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.4\\variants\\esp32\\partitions.csv" "C:\\Users\\user1\\AppData\\Local\\Temp\\arduino_build_167513\\partitions.csv"

cmd /c if not exist "C:\\Users\\user1\\AppData\\Local\\Temp\\arduino_build_167513\\partitions.csv" COPY "C:\\Users\\user1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.4\\tools\\partitions\\default.csv" "C:\\Users\\user1\\AppData\\Local\\Temp\\arduino_build_167513\\partitions.csv"

cmd /c IF EXIST "C:\\Users\\user1\\OneDrive\\Documentos\\Arduino\\sketch_sep15a\\sketch_sep15a\\bootloader.bin" ( COPY /y "C:\\Users\\user1\\OneDrive\\Documentos\\Arduino\\sketch_sep15a\\sketch_sep15a\\bootloader.bin" "C:\\Users\\user1\\AppData\\Local\\Temp\\arduino_build_167513\\sketch_sep15a.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\user1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.4\\variants\\esp32\\bootloader.bin" ( COPY "C:\\Users\\user1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.4\\variants\\esp32\\bootloader.bin" "C:\\Users\\user1\\AppData\\Local\\Temp\\arduino_build_167513\\ [... cut ...]
                 from C:\Users\user1\OneDrive\Documentos\Arduino\sketch_sep15a\sketch_sep15a\sketch_sep15a.ino:2:

C:\Users\user1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32/include/hal/include/hal/dac_types.h:29:3: note: previous declaration as 'typedef enum dac_cw_phase_t dac_cw_phase_t'

 } dac_cw_phase_t;

   ^~~~~~~~~~~~~~

Using library DacESP32 at version 1.0.9 in folder: C:\Users\user1\Documentos\Arduino\libraries\DacESP32 

exit status 1

Error compiling for board ESP32 Dev Module.

yellobyte, 2022-09-15:
Hi there,
looks like you don't use the official Espressif dev package. What settings do you use ?
Make sure you have added "https://dl.espressif.com/dl/package_esp32_index.json" to Settings->Additional Board Manager URLs.  Then under Tools->Boards Manager  select the newest Espressif ESP32 package "V1.0.6". See attached pics.
Just did a build with these settings and building "outputDigitalAnalogMixed" finished without errors.
Come back to me if you still have problems. Thanks Thomas   

[image]
   

[image]

Fabgar1, 2022-09-15:
Thank you so much, I was using the dev package from  https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json that I supposed to be the official.
And thanks again from the prompt answer!

yellobyte, 2022-09-15:
Just installed Arduino package 2.0.4 (json from some other source) and can replicate your error. Sorry, seems I was a bit behind time....  Will prepare DacESP32 version 1.0.10 and come back to you.

Fabgar1, 2022-09-15:
Man, not so behind, I was on 1.0.6 until yesterday....

yellobyte, 2022-09-15:
Ok, I put Release V1.0.10 online. Please confirm a successful build with your IDE settings. Thanks Thomas

Fabgar1, 2022-09-16:
Working! Thank you so much!

yellobyte, 2022-09-16:
Cheers. If you like the lib, please give it a star. Thanks Thomas

closed 2022-09-16 by yellobyte