yellobyte/DacESP32 #2: Multiple variables not declared inside the library

makmal0962, 2023-01-14:
I tried to compile the example "outputCW" to my DOIT ESP32 DEVKIT V1 using Arduino IDE 2.0.3 and i got this compiler error.

What i'm assumming is that there are so many declaration errors inside the library, that also affect the example sketch.

The compiler output:
`Using board 'esp32doit-devkit-v1' from platform in folder: C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "C:\\Users\\LENOVO\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023014-18972-xmj378.a676h\\outputCW\\partitions.csv" copy /y "C:\\Users\\LENOVO\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023014-18972-xmj378.a676h\\outputCW\\partitions.csv" "C:\\Users\\LENOVO\\AppData\\Local\\Temp\\arduino-sketch-C7A7085FB720561C6FCCE43CB6C9CE93\\partitions.csv"
cmd /c if not exist "C:\\Users\\LENOVO\\AppData\\Local\\Temp\\arduino-sketch-C7A7085FB720561C6FCCE43CB6C9CE93\\partitions.csv" copy "C:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\LENOVO\\AppData\\Local\\Temp\\arduino-sketch-C7A7085FB720561C6FCCE43CB6C9CE93\\partitions.csv"
        1 file(s) copied.
Detecting libraries used...
"C:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/expat" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/fatfs" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freemodbus" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freertos" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/heap" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/idf_test" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/jsmn" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/json" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/libsodium" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/log" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/lwip" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mbedtls" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mdns" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/micro-ecc" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mqtt" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/newlib" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nghttp" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nvs_flash" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/openssl" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/protobuf-c" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/protocomm" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/pthread" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/sdmmc" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/soc" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/spi_flash" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/spiffs" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/tcp_transport" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/tcpip_adapter" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ulp" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/unity" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/vfs" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/wear_levelling" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/wifi_provisioning" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/wpa_supplicant" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\LENOVO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-face" "-IC:\\Users\\L [... cut ...]
C:\Users\LENOVO\AppData\Local\Temp\.arduinoIDE-unsaved2023014-18972-xmj378.a676h\outputCW\outputCW.ino:40:23: error: 'DAC_CW_SCALE_4' was not declared in this scope
       dac1.setCwScale(DAC_CW_SCALE_4);
                       ^
C:\Users\LENOVO\AppData\Local\Temp\.arduinoIDE-unsaved2023014-18972-xmj378.a676h\outputCW\outputCW.ino:42:23: error: 'DAC_CW_SCALE_8' was not declared in this scope
       dac1.setCwScale(DAC_CW_SCALE_8);
                       ^

Using library DacESP32 at version 1.0.10 in folder: C:\Users\LENOVO\OneDrive\Documents\Arduino\libraries\DacESP32 
exit status 1

Compilation error: 'DAC_CW_SCALE_1' was not declared in this scope`

makmal0962, 2023-01-14:
resolved because I didn't read the Readme before.

yellobyte, 2023-01-14:
No worries. Glad you found it yourself... :-)
Thanks for using the lib. If you like it, please give it a star. 
Regards Thomas

closed 2023-01-14 by yellobyte