yellobyte/DacESP32 #4: function naming is ambiguous

quick0831, 2023-02-18:
Just some suggestions in function naming
I find the usage ambiguous (the compiler doesn't like it as well)
```cpp
esp_err_t outputVoltage(uint8_t value);
esp_err_t outputVoltage(float voltage);
```
[image]
The compiler still prefer double over float

maybe change it to:
```cpp
esp_err_t outputValue(uint8_t value);
esp_err_t outputVoltage(float voltage);
```

yellobyte, 2023-02-18:
Hi there, calling an overloaded function without typecasting the argument(s) passed is generally not a good idea. The compiler might speculate right, often not. In latter case the problem can stay unnoticed esp. without compile error. 
Below line will compile without hassle and delivers the result you want:
   ...
   **dac.outputVoltage((float)(sin(f) * 1.5 + 1.6));**
   ...
When coding don't let the compiler guessing. Try to make sure the argument is interpreted properly and all participants are happy. Regards Th

yellobyte, 2023-02-18:
Just as a further remark, the floating argument passed to outputVoltage() is supposed to be in the range 0...CHANNEL_VOLTAGE_MAX (3.3 by default). An argument with higher value than CHANNEL_VOLTAGE_MAX is accepted though but always results in max possible output voltage.

quick0831, 2023-02-18:
The reason I found it ambiguous is that the overload both accept one numerical argument, however they accept arguments with different meaning. (One is byte range, the other is actual voltage) Directly naming the byte range one `outputVoltage` might not be a good idea since it's not a voltage, so I want to change it into a more declarative name.
(Sorry for not being clear enough on the issue description)

On the other hand, I accept the point that calling overloaded function without type casting is a bad idea, and changing function name is a breaking change, which might be a worse idea.

yellobyte, 2023-02-18:
Well, I get your point. I named it outputVoltage() because arguments of both types will finally result in a certain "output voltage" level, represented by the functions argument. I guess using float makes it easier for humans to recognize/read without having to calculate uint8Value / 255 * 3.3V.
Applying a typecast in code however is a very precise and declarative way to say what you want and what the argument represents. Maybe that helps...

quick0831, 2023-02-19:
Yeah. Maybe I'll close the issue?

yellobyte, 2023-02-19:
Just done. Thanks for the discussion. Regards Th

closed 2023-02-19 by yellobyte