yellobyte/DacESP32 #7: hal/dac_types.h: No such file or directory

Rocaj74, 2025-01-26:
Dear yellobyte, 
I am trying to use the DacESP32 library on my ESP32-Wroom-32 board, but I have the following error (for instance with the example  outputCW.ino): **dac_types.h: No such file or directory**

Here the whole error message: 
In file included from C:\Users\Root\AppData\Local\Temp\.arduinoIDE-unsaved2025026-15224-jhg8mx.ql1xd\outputCW\outputCW.ino:14:0:
c:\Users\Root\Documents\Arduino2\libraries\DacESP32\src/DacESP32.h:47:27: fatal error: hal/dac_types.h: No such file or directory
compilation terminated.

I am using Arduinio IDE 2.2.1
Thank you very much, 
Roberto.

yellobyte, 2025-01-26:
what board/CPU are you exactly building for ?

yellobyte, 2025-01-26:
[image]

Rocaj74, 2025-01-29:
Dear yellobyte, thank you for the answer. As board release (ESP32 By Espressif Systems) I am using the old **v1.0.6**, so I guess this is the problem! Unfortunately at the moment I cannot update it, since the newer versions are not compatible with other projects i am working with! At the moment I found a workaroud generating a PWM signal and filtering it with a Low Pass filer. 
Eventually I will try on another PC (to avoid messing up the old libraries/board release). 
FYI, I am using a board that should be compatible with ESP32, this one below. 
Thank you again for the help, 
Roberto. 

[image]

yellobyte, 2025-01-29:
Hi Roberto,
DacESP32 lib v1.x.x needs Arduino ESP32 v2.x and lib v2.x.x needs at least ESP32 v3.x. 
Arduino ESP32 v1.0.x came out in 2021 already and is slightly deprecated, you should start thinking about an upgrade...
Regards Th