yellobyte/DacESP32 #8: loop() gets permanently slowed down after using dac2

CRSourcing, 2025-03-05:
Hello,

Arduino IDE 2.3.4 with ESP32 IDF 3.1.3.  
The main loop() gets permanently slowed down after I run this function just once: 

void sineTone(uint32_t frequency, uint32_t duration) {  // sine wave generator for touch sound, frequency =  440Hz, duration = 20ms. 
dac2.enable();
dac2.outputCW(frequency);
delay(duration);
dac2.disable();
}

loop(), which runs several tasks, needs 15ms before calling this function and about 30ms after calling it.

yellobyte, 2025-03-06:
Hi there,
it's hard to tell what causes this with this little info provided. The lib sits on top of Espressifs ESP32 v3.x new & re-designed dac drivers.
(https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/migration-guides/release-5.x/5.1/peripherals.html)
 Questions:
1) what build options are used and has building with/without CW_FREQUENCY_HIGH_ACCURACY any effect ?
2) how do you measure & print the looptime ?
3) what else does your program in loop(), mind to share your source code ?
4) where is dac2 defined, within function loop() or globally ?  Is there a dac1 defined and in use ?
5) what is your IDE ? Version ? What esp32 framework version do you use ? Pls provide your build log.

You can perform some tests and check the looptimes:
a) remove _dac2.enable();_ It is **not** needed here, outputCW() does it for you
b) change sineTone() as below
```
void sineTone(uint32_t frequency, uint32_t duration) {
  //dac2.enable();  <-- not needed
  dac2.outputCW(frequency);
  delay(duration);
  dac2.disable();
}
```
c) and test again with
```
void sineTone(uint32_t frequency, uint32_t duration) {
  dac2.enable();
  //dac2.outputCW(frequency);
  //delay(duration);
  dac2.disable();
}
```
Pls come back with your findings.
Regards yb

CRSourcing, 2025-03-06:
Hello yellobyte,
Thank you for looking into this.  Please do not invest too much time. I am using dac2 to provide a short acoustic confirmation if a touchscreen gets touched, but I can also use a square wave to do this.
 
To answer your questions:

//dac2.enable();  <-- not needed 
-Commenting it out does not change anything.

what build options are used and has building with/without CW_FREQUENCY_HIGH_ACCURACY any effect ? 
- No additional options and building without  CW_FREQUENCY_HIGH_ACCURACY does not change anything. 

how do you measure & print the looptime ? 
- loop() includes a looptime measurement tool. Results are displayed on the tft display. 
Looptime increases permanently from 15ms to 31ms after calling sineTone().  Controls feel sluggish. 

what else does your program in loop(), mind to share your source code ? 
- loop() runs a list of tasks related to a receiver. The ESP32 drives a receiver chip, a tft touchscreen and peripherals. I2C and SPI bus are constantly in use. Code:

void loop() {

  encoderMoved();         // check whether encoder has moved
  displaySTEP();          // check STEP
  fineTune();             // read frequency potentiometer
  saveCurrentSettings();  // save settings after 1 minute if freq has not changed

  if (FREQ != FREQ_OLD) {  // If FREQ changes, update Si5351A
    FREQCheck();           //check whether within FREQ range
    displayFREQ(FREQ);     // display new FREQ
    tuneSI5351();          // and tune it in
    autoloopBands();       // if autoloop is enabled frequency will stay within a selected band
    FREQ_OLD = FREQ;
  }

  // functions that don't have to run in every loop cycle get called with fTrigger

  if (fTrigger % 3 == 0) {
    readTouchCoordinates();  // this function takes long. Calling it 1 out of 3 loop cycles is sufficient.
    readSquelchPot(true);    // true = read and draw position circle
    setSquelch();
    setMode();     //  set mode (tune or scan mode) must run after readTouchCoordinates(). Needs a recent value of pressed
    touchTuner();  // graphical frequency display, switches to FREQ when touched
    mainScreen();  // build main window
  }

  if (fTrigger % 8 == 0 && disableFFT == false) {  // spectrum eats a lot of processing resources
    audioSpectrum();
  }

#ifdef TINYSA_PRESENT
  if (fTrigger % 15 == 0)
    synctinySA();  // tinySA synchronisation when in RF mode
#endif

  if (fTrigger % 25 == 0) {
    clockDisplay();  // clock in lower left corner
    if (setFilterManually)
      manualFilterSelector();
  }

  if (fTrigger % 50 == 0) {

    if (syncEnabled == false || useTSAForSmeter == false)
      Smeter();         // Smeter about 3x second to reduce noise on I2C, alternatively gets called from convertToDbm(),when 
  tinySA is the source
    slowTaskHandler();  // run tasks with a long period
    displayFREQ(FREQ);  // update in case clock source or injection mode has changed
    fTrigger = 0;
    disableFFT = false;
  }

  fTrigger++;  // function trigger counter, triggers functions that should not run every cycle
}
 
 
where is dac2 defined, within function loop() or globally ? Is there a dac1 defined and in use ? 
- dac2 globally and no dac1 defined or in use. GPIO25 is in use, but not in dac mode.

pls provide your build log 
- No particular build log (Arduino IDE), but output log is clean (no warnings).

what esp32 board do you use ? -
-ESP32 devkit, generic

yellobyte, 2025-03-06:
Hi,
I'm now curious as to what causes the delay. If you dont mind, please test the 2 following scenarios and check the respective looptimes.
1) Add a file named "_build_opt.h_" to your sketch directory, containing the line
`-DCW_FREQUENCY_HIGH_ACCURACY=0`
2) Change sineTone() to 
```
void sineTone(uint32_t frequency, uint32_t duration) {
  dac2.enable();
  dac2.disable();
}
```
Thanks yb

CRSourcing, 2025-03-06:
Good news, 
Adding a file named "build_opt.h" to the sketch directory, containing the line
-DCW_FREQUENCY_HIGH_ACCURACY=0  
 resolved the issue.

Thanks for your help.

yellobyte, 2025-03-07:
I just [added a few lines to the build options section in the Readme file](https://github.com/yellobyte/DacESP32?tab=readme-ov-file#hammer_and_wrench-available-library-build-options) referring to that situation in case other users stumble across this particular problem as well. Thanks for your contribution!
Regards Thomas

closed 2025-03-07 by yellobyte