yellobyte/DacESP32 #9: setCwFrequency causes drop-outs in DAC signal

eslantura, 2025-04-14:
First of all, Thomas thank you for the great job!

I've encountered room for improvement when it comes to change the CW frequency, like used with frequnecy sweeps, for example.
Unregistering and registering a DAC channel causes drop-outs in the DAC signal. If two channels are used, only the instance requesting the frequency set is touched. So, the idea has come up to just change the frequency registers, rather than to delete and open a new handle.
I've successfully tried it by using a copy of the calcFrequSettings function.
It would be desireable if you could extend the library by a function which can change the CW frequency "on-the-fly", at least if CW_FREQUENCY_HIGH_ACCURACY is set.

Thanks and regards,
Peter F.

yellobyte, 2025-04-16:
Hi Peter, thanks for the feedback. I'll have a look at it when time permits and come back to you. Regards yb

eslantura, 2025-04-16:
Hi Thomas, Great! I think a static function referenced inside the class would be preferable. Regards, Peter

yellobyte, 2025-04-20:
New lib version 2.1.0 has been uploaded. Added 2 new examples: outputCWfastFrequSweep.ino and outputCWfastFrequSweep2.ino.
Have a look at outputCWfastFrequSweep.jpg and outputCWfastFrequSweep2.jpg in folder doc.
Pls give the lib a decent check. Regards yb

eslantura, 2025-04-27:
Hi Thomas, Thanks a lot. I've integrated the new version and tested it. It works perfectly.
Regards, Peter

yellobyte, 2025-04-27:
Thanks Peter for testing, I just created the new release v2.1.0. Regards yb

closed 2025-04-27 by yellobyte